oemof/feedinlib #9: Apply inverter model

gplssm, 2015-10-13:
PVLIB provides an [inverter model](http://pvlib-python.readthedocs.org/en/latest/pvlib.html?highlight=inverter#pvlib.pvsystem.snlinverter). Inverter can be retrieved [from databsae](http://pvlib-python.readthedocs.org/en/latest/pvlib.html?highlight=inverter#pvlib.pvsystem.retrieve_sam).

gplssm, 2015-11-18:
deferred to 0.0.9 and maybe won't be implemented here due to overlap with pvlib (see #16). If [this PR](https://github.com/pvlib/pvlib-python/pull/93) is merged, system description is already done by pvlib.

uvchik, 2016-02-15:
I thought you want to have the output of the inverter and not of the module. How is it done with [PR#93](../../../pvlib/pvlib-python/pull/93) of the the pvlib?

wholmgren, 2016-02-15:
Here's a link that PR's new documentation since I think that it might be lost in the long thread:

http://wholmgren-pvlib-python-new.readthedocs.org/en/pvsystem/package_overview.html#modeling-paradigms

I haven't had much substantiative feedback on #93, which makes me think that people love it (unlikely), people don't care at all (maybe), or don't care to get involved until their existing code breaks (likely). Would be happy to hear your thoughts on it.

closed 2018-01-17 by birgits