oemof/feedinlib #20: Duplication of powerplant classes in feedinlib

cswh, 2016-01-26:
@oemof/feedinlib , @oemof/oemof-main 
So far the powerplant classes (Base and its children) in powerplants.py duplicate the powerplant classes idea of oemof_base. Either it should be a child class of an oemof_base component class (and then it should be integrated in oemof_base, because it is not independent anymore) or feedinlib should consists more of just functions and no powerplant classes anymore.

What do you think?

uvchik, 2016-01-26:
I just kept your structure to avoid discussions. So we can also discuss a new structure. 

Any suggestions?

ckaldemeyer, 2016-01-27:
What speaks against an integration into oemof_base?

ckaldemeyer, 2016-01-27:
As we need feedin-series in almost every model..

uvchik, 2016-01-27:
What speaks against just using it within oemof_base?

So that other people still can use it as a stand-alone application.

ckaldemeyer, 2016-01-29:
> What speaks against just using it within oemof_base?
> So that other people still can use it as a stand-alone application.

But if I need a package for a certain purpose in most cases the package has many other features. Let's say I only want to do matrix operations and use scipy including numpy. Besides the matrix operations scipy has many other features such as solving linear equations, ...

uvchik, 2016-01-29:
If you want to use [numpy](https://pypi.python.org/pypi/numpy) you do not have to install [scipy](https://pypi.python.org/pypi/scipy/0.7.0) but scipy has numpy as a requirement.

I know that there are arguments for both ways. I agree that you can use it either way but I think it lowers the entry barrier for contributors. The oemof package is quite complex and it might deter people from contributing if they just want to improve the wind model.

Does it annoy you in your daily work if we keep it as two packages?

@cswh , @gnn and I planing to integrate the feedinlib better into oemof but keep it as two packages. Maybe you could have a look at the changes and if still does not work good for you we can discuss joining it again.

Would it help you if we create the same name space as @gnn explained it for oemof_pg [(see issue 45 of oemof_base)](https://github.com/oemof/oemof_base/issues/45#issuecomment-176376199)?

ckaldemeyer, 2016-01-29:
> I know that there are arguments for both ways. I agree that you can use it either way but I think it lowers the entry barrier for contributors. The oemof package is quite complex and it might deter people from contributing if they just want to improve the wind model.

> Does it annoy you in your daily work if we keep it as two packages?

No, it doesn't and I also think that one can argue both ways. For me it is only important that it is 100% compatible and if possible without (slightly) redundant structures. But I don't have time and I gues the ones who do stuff (you and @cswh ) can decide.

uvchik, 2016-02-15:
@cswh : What is the time table for these changes? Proposal, implementation, testing,...

cswh, 2016-03-01:
@gnn will give an update here soon.

gnn, 2016-03-01:
Sorry for having this dormant for so long. I'll code a proposal for a new structure (hopefully at least a week) before 2016-04-06 so that we have a basis for discussion during the refactoring meeting.

uvchik, 2016-03-01:
According to the results of the web-conference it could be better to post a concept instead of producing code. 

As I understand the discussion there are different opinions considering the idea of a framework.

gnn, 2016-03-01:
Thanks for the input. To clarify: the code I will (hopefully) produce is supposed to illustrate the concept on how `feedinlib` can look like and be used in the future. It can then serve as a concrete example on which we can base the later discussion. If the proposal is rejected during that discussion, that's fine.

uvchik, 2016-03-02:
Cool :smile: I am looking forward to your proposal.

gnn, 2016-08-26:
Alright. Finally coming back to this. Hopefully I'll get this done before the milestone.
In order to get this done quickly and keep working on it consistently, I'll post an update here next Friday, regardless of how far I've gotten.

closed 2018-01-17 by birgits