oemof/feedinlib #21: Part of feedinlib depends on oemof_test schema in oemof database

cswh, 2016-01-26:
@oemof/feedinlib 
As I understood, the main reason to have a separate feedinlib (not included in the initial oemof rep) was that everything in feedinlib should be independent from the rest of oemof. In the class SimpleWindTurbine the method fetch_cp_values_from_db refers to the oemof database. In my opinion it should not be in such a generic structure of feedinlib.

Is that just a leftover from the intial code?

uvchik, 2016-01-26:
You are right. we could remove this method and the according code. Will you do it?

Be aware that you also have to remove the call of the function [call of the function](../blob/master/feedinlib/models.py#L590) and the "conn" parameter.

uvchik, 2016-02-15:
I removed it.

closed 2016-02-15 by uvchik