oemof/feedinlib #27: Solar Thermal Energy Modelling?

simnh, 2017-01-08:
Does a library exist to model solar thermal energy? If, not did you model that already in combination with oemof (or standalone) ? 

Thx!

birgits, 2017-01-09:
There is no python model for this. We modelled the solarthermal hot water production for different setups (different orientations of the collectors, different heat demands, warm water production and supplementary heating support) with TRNSYS a while ago and are simply scaling those feedin timeseries in our apps. If you want I can send them to you.

simnh, 2017-01-09:
Ok, sound good, but ow did you scale? With weather data ?

uvchik, 2017-01-09:
We scaled with square meters.

I do have some octave/matlab code and I planned to translate it but did not manage to do it.

It really depends on what you are going to do. For some tasks my code might be interesting for other tasks the pre-simulation with e.g. TRNSYS might be better.

uvchik, 2017-01-09:
You have to do the TRNSYS simulation with your weather data.

simnh, 2017-01-09:
> You have to do the TRNSYS simulation with your weather data.

Now this makes sense, so for my region, that would possibly make now sense to use other TRNSYS data. 
I was rather looking for something like the feedin lib...

I want to use it in a model for Schleswig Holstein (oemof -investment). So basiccally I am looking for a normed thermal heat production curve from solar thermal power. I would just ignore temperature levels for now. and say heat is heat...which of course  is not the case.

uvchik, 2017-01-09:
Solar thermal systems for district heat or domestic heat?

simnh, 2017-01-09:
Ideally both, but one of both would be better than nothing of course ;)

c-moeller, 2020-02-19:
Solar thermal components are now part of oemof.thermal and TESPY. I'm closing this issue.

closed 2020-02-19 by c-moeller