oemof/feedinlib #28: webco about the future of the feedinlib

birgits, 2017-05-29:
@gnn and @uvchik it would be good to have a webco where we discuss the future of the feedinlib soon. Would any time next week on Wednesday, Thursday or Friday work for you?

uvchik, 2017-06-07:
From 10 to 15 and from 16.30 to 18.30 on Wednesday, Thursday or Friday.

birgits, 2017-06-07:
@uvchik I just talked to gnn on the phone about a webco date. Would next Wednesday afternoon work for you?

uvchik, 2017-06-09:
Okay, Wednesday from 14.00 to 15.00 will be okay. If we need more time we can meet again in the later afternoon, but I guess one hour is far enough.

birgits, 2017-06-14:
Links from webco:
https://www.researchgate.net/post/Does_any_one_have_a_script_in_r_or_python_to_pull_Merra_data
https://github.com/Unidata/netcdf4-python
http://www.geomar.de/mitarbeiter/fb1/me/jharlass/wrf/erai/

gnn, 2017-06-16:
These are the minutes I wrote down during our WebCo:

#### 2017-06-14: Feedinlib quo vadis

  - Einheitliches Wetterobjekt
    - Einheitliche/Standardisierte Namen für Messreien
    - PVlib/Windpowerlib: Dataframes
  - Anschluß an Wetterdatensatz
  - Einheitliche Schnittstelle für techn. spezialisierte Bibliotheken (e.g.: pvlib, windpowerlib)
  - JanUrb? MERRA online API?

They're only here for historical reasons as I translated them to English and converted them to the roadmap written down in #29.

birgits, 2017-08-18:
@gnn did we decide to close this issue? I'm not sure if you wanted to add some documentation before closing it.

closed 2018-01-17 by birgits