oemof/feedinlib #30: Weather data

birgits, 2017-07-21:
This issue serves as a reminder on what we discussed and found on weather data so far.

Open access data we could provide download scripts for are:
* MERRA2
* ERA5
* ERAI

There are already scripts for MERRA2 and ERAI that we could use:
* MERRA2 - https://github.com/Open-Power-System-Data/weather_data
* ERAI - http://www.geomar.de/mitarbeiter/fb1/me/jharlass/wrf/erai/

birgits, 2018-01-17:
Another radiation data set that was mentioned in #15 is [WRMC-BSRN](http://bsrn.awi.de/). It is a free access data archive of measured meteorological data.

birgits, 2018-02-21:
COSMO-REA6 is another free weather data set, licensed under GeoNutzV and available for download here: http://reanalysis.meteo.uni-bonn.de/?Download_Data___COSMO-REA6

Ludee, 2018-02-21:
I can add the [JRC weather data (PVGIS)](http://re.jrc.ec.europa.eu/pvgis.html) to your list.

birgits, 2019-12-17:
@Pyosch offered to add DWD weather data :)

Pyosch, 2019-12-20:
The pull request for the script is [here](https://github.com/oemof/feedinlib/pull/42). Maybe you could give me some feedback, in what form and where you would like to include it :)