oemof/feedinlib #47: ConnectionException for openFRED location table

c-koehl, 2020-05-06:
Hey,
I'm trying to read the openFRED data:
```python
from feedinlib.open_FRED import Weather
from feedinlib.open_FRED import defaultdb
from shapely.geometry import Point, Polygon

lat = 52.4
lon = 13.5
location = Point(lon, lat)
open_FRED_weather_data = Weather(
    start='2017-06-01', stop='2017-07-01',
    locations=[location],
    variables="pvlib",
    **defaultdb())
```
but somehow I receive a connection exception (even though I used my token):

`oedialect.engine.ConnectionException: HTTP 400 (Bad Request): Table sandbox.openfred_locations not found`

But if I run following code, I can get access to the location`s data:

```python
import oedialect
from sqlalchemy import Table, create_engine, MetaData, Column
from sqlalchemy.orm import sessionmaker

engine = create_engine('postgresql+oedialect://MYUSERNAME:MYTOKEN@openenergy-platform.org')

Session = sessionmaker()
session = Session(bind=engine)

metadata = MetaData(bind=engine)

loc = Table("openfred_locations", metadata,
    Column("id"),
    Column("point"),
    schema="climate")

q = session.query(loc).limit(10)
```

So it doesn't seem to be a database bug. Can you fix this problem or am I doing something wrong? I was trying to find the bug, but I'm quite new to database structures, so I wasn't able to find anything.
Thanks!

P.S.: Here is the entire error message:
```
Traceback (most recent call last):
  File "/Volumes/Consolinno/feedinlib/test_fred.py", line 60, in <module>
    **defaultdb())
  File "/Volumes/Consolinno/feedinlib/pvenv/lib/python3.7/site-packages/feedinlib/open_FRED.py", line 138, in __init__
    if locations is not None
  File "/Volumes/Consolinno/feedinlib/pvenv/lib/python3.7/site-packages/feedinlib/open_FRED.py", line 137, in <dictcomp>
    {(l.x, l.y): self.location(l) for l in locations}
  File "/Volumes/Consolinno/feedinlib/pvenv/lib/python3.7/site-packages/feedinlib/open_FRED.py", line 307, in location
    .order_by(self.db["Location"].point.distance_centroid(point))
  File "/Volumes/Consolinno/feedinlib/pvenv/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 3300, in first
    ret = list(self[0:1])
  File "/Volumes/Consolinno/feedinlib/pvenv/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 3078, in __getitem__
    return list(res)
  File "/Volumes/Consolinno/feedinlib/pvenv/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 3405, in __iter__
    return self._execute_and_instances(context)
  File "/Volumes/Consolinno/feedinlib/pvenv/lib/python3.7/site-packages/sqlalchemy/orm/query.py", line 3430, in _execute_and_instances
    result = conn.execute(querycontext.statement, self._params)
  File "/Volumes/Consolinno/feedinlib/pvenv/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 984, in execute
    return meth(self, multiparams, params)
  File "/Volumes/Consolinno/feedinlib/pvenv/lib/python3.7/site-packages/sqlalchemy/sql/elements.py", line 293, in _execute_on_connection
    return connection._execute_clauseelement(self, multiparams, params)
  File "/Volumes/Consolinno/feedinlib/pvenv/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1103, in _execute_clauseelement
    distilled_params,
  File "/Volumes/Consolinno/feedinlib/pvenv/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1288, in _execute_context
    e, statement, parameters, cursor, context
  File "/Volumes/Consolinno/feedinlib/pvenv/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1485, in _handle_dbapi_exception
    util.raise_(exc_info[1], with_traceback=exc_info[2])
  File "/Volumes/Consolinno/feedinlib/pvenv/lib/python3.7/site-packages/sqlalchemy/util/compat.py", line 178, in raise_
    raise exception
  File "/Volumes/Consolinno/feedinlib/pvenv/lib/python3.7/site-packages/sqlalchemy/engine/base.py", line 1248, in _execute_context
    cursor, statement, parameters, context
  File "/Volumes/Consolinno/feedinlib/pvenv/lib/python3.7/site-packages/sqlalchemy/engine/default.py", line 590, in do_execute
    cursor.execute(statement, parameters)
  File "/Volumes/Consolinno/feedinlib/pvenv/lib/python3.7/site-packages/oedialect/engine.py", line 338, in execute
    requires_connection_id=requires_connection_id)
  File "/Volumes/Consolinno/feedinlib/pvenv/lib/python3.7/site-packages/oedialect/engine.py", line 355, in __execute_by_post
    requires_connection_id=requires_connection_id)
  File "/Volumes/Consolinno/feedinlib/pvenv/lib/python3.7/site-packages/oedialect/engine.py", line 232, in post
    process_returntype(ans, json_response)
  File "/Volumes/Consolinno/feedinlib/pvenv/lib/python3.7/site-packages/oedialect/engine.py", line 241, in process_returntype
    raise ConnectionException('HTTP %d (%s): %s'%(response.status_code,response.reason, message))
oedialect.engine.ConnectionException: HTTP 400 (Bad Request): Table sandbox.openfred_locations not found
```

gnn, 2020-05-13:
Thanks for the report. Are you installing the package from source? If so, based on which branch?
And then there's something in the output which doesn't match up with what I expected. I have to look into that.

c-koehl, 2020-05-14:
I cloned the repository and installed the dependencies.
I remember having issues installing it via pip (3 weeks ago). Apparently if you install it via pip, an older version will be installed, because some files were missing and were different compared to the dev branch in github!

nick-harder, 2020-07-22:
Hi, is there a solution for this issue already? I have tried running it on my PC and also on Binder, it doesn't work

p-snft, 2020-11-12:
I can reproduce the issue as reported by @nick-harder, also for the current dev. However, I do not have the appropriate knowledga about open_FRED, to fix it. @oemof/feedinlib, could anyone of you help?

gnn, 2020-11-12:
Sorry for taking so long to fix this and thanks for continuously prodding me. The issue that @nick-harder experienced and which @p-snft reproduced is different from the one originally reported. I'll try to reproduce both and probably will open a new issue for the former. In the long run, I'm hoping to have both issues fixed in a month.

ADai528, 2020-12-08:
hi, @nick-harder. I also encountered the same problem, did you finally solve this problem?