oemof/feedinlib #48: Link in README to examples section is broken

fwitte, 2020-05-27:
This link does not work:
`.. _examples: https://feedinlib.readthedocs.io/en/stable/examples.html`
I did not know where it should point to, so maybe someone can fix this.

birgits, 2020-05-28:
Thanks for reporting this @fwitte. When I set this link I thought we would release soon, but as we only released release candidates since the restructuring, there is no examples section on the master branch. Therefore, it should be changed to `latest`, but unfortunately that branch currently doesn't compile on readthedocs... I will fix it when I find the time. For now if you are interested you may use this link:
https://feedinlib.readthedocs.io/en/releases-0.1.0/examples.html

uvchik, 2021-06-22:
We should check all links before the release of v0.1.0