oemof/feedinlib #49: Allow for stochastic demand profiles

p-snft, 2020-07-15:
Standard lows profiles (SLP) are – by construction – static. This can be a problem especially for electric demands of small sites (like single family homes). Thus, I suggest to introduce an option to get some randomness:

* Part of the SLP is considered base load (constant over time).
* The rest is composed of a static and a dynamic part.
    * The static part scales with the original SLP. Thus, setting this to 1 will just reproduce the SLP.
    * The dynamic part is modelled by a  Poisson distribution with an expectation value scaled with the original SLP.

So – by construction –  the SLP is reconstructed on average. Of course, this simple model is not very realistic, as it does not allow correlations between following values. However, it should be better then plain SLP.

birgits, 2020-07-16:
@p-snft did you mean to post this in the demandlib?

p-snft, 2020-07-30:
Oops, yes. Are you able to move it?

closed 2020-07-31 by p-snft