oemof/feedinlib #53: can not run file [load_open_fred_weather_data.ipynb] online by binder or using my PC

ADai528, 2020-12-02:
hi, I want to run load_open_fred_weather_data.ipynb file, but no matter I use binder online or on my PC , errors will appear.

When i use the binder online, it shows that the **'open_FRED' module cannot be found**.

[image]

When i try to run it on my PC, there will be another error: **AttributeError: 'NoneType' object has no attribute 'id'**

[image]
[image]

**Has anyone encountered the same problem and how did you solve it?  I hope that feedinlib Team **can** solve this problem as soon as possible.**

dpinney, 2021-01-10:
I got a little farther by adding an initialization cell:

```
!pip install "git+https://github.com/oemof/feedinlib.git"
!pip install open_FRED-cli
```

But then hit:

[image]

dpinney, 2021-01-10:
I gave up on FRED and switched to looking at era5. Looks like there's some configuration steps needed to get access to the api and to store the key.

dpinney, 2021-01-10:
Okay, I finally got this all running with era5 data. Here's a working notebook: https://colab.research.google.com/drive/174mEpFy-Yr00o3R2Grqd8Q7mvGsG-jHY

Note you'll have to go through the era5 api signup and key generation process to get it to run.

I'd recommend updating the feedinlib readme and install instructions to mention that registrations and api keys are needed for the weather data sources (I assume FRED wasn't working because I didn't register).

It also appears that an old version of feedinlib is in pypi that's incompatible with the example notebooks. That's why I did a `!pip install git+https://github...` install.

By the way, great job in the library finding data and a preparation pipeline that can be used as an input to windpowerlib. Not an easy task!

uvchik, 2021-06-22:
Thank you @dpinney for sharing your experience. We will check the notebooks before the next release and we will have a look at your notebook.