oemof/feedinlib #57: Bug when using feedinlib in combination with pvlib

GregorBecker, 2021-04-07:
When using "feedinlib version 0.0.12" with "pvlib0.7.1" the following error occurs (see attachment).
Does it come to version competition?
[image]

uvchik, 2021-06-21:
Sorry the project got stuck in the last time but we are working hard to release a new version soon. You can use the `dev` branch but I would recommend to merge the feature branches.

uvchik, 2021-06-22:
We should write test to check the usage of the pvlib. This is related to #62.

GregorBecker, 2021-06-23:
Thank you for the response @uvchik, we will test it and are looking forward to the new version.