oemof/feedinlib #62: Deprecated warning with pvlib

uvchik, 2021-06-17:
If we are not able to fix that soon, we should restrict the version of the pvlib to `<0.9`.

```
.../site-packages/pvlib/pvsystem.py:485: pvlibDeprecationWarning: 

temperature_model_parameters, racking_model, and module_type are not specified. Reverting to deprecated default: SAPM cell temperature model parameters for a glass/glass module in open racking.
In v0.9, temperature_model_parameters or a valid combination of racking_model and module_type will be required.

pvlibDeprecationWarning
```

uvchik, 2021-06-17:
I will be responsible for this Issue, but I do not have time to fix it, so please help.

p-snft, 2021-11-11:
I would rather not depend on an outdated version of anything. Instead, I'd rather remove the functionality. If pvlib requires too many details, we might switch to a simple [geometric model](https://github.com/oemof/feedinlib/blob/dev/src/feedinlib/models/geometric_solar.py). (I think that already picking default modules and inverters is somewhat unclean.)