oemof/feedinlib #71: Incomplete input data. Data needs to contain ['ghi', 'dni', 'dhi']. Detected data contains: ['wind_speed', 'temp_air', 'ghi', 'dhi']

lindaken, 2022-02-18:
Cannot convert pv_lib to image, always missing "dni".
[image]

birgits, 2022-02-28:
The pvlib requires you to provide DNI in your weather data as well. You can use [this function](https://github.com/pvlib/pvlib-python/blob/master/pvlib/irradiance.py#L2849) to calculate the DNI. Does that solve your problem?

jenny-nyx, 2025-07-14:
> The pvlib requires you to provide DNI in your weather data as well. You can use [this function](https://github.com/pvlib/pvlib-python/blob/master/pvlib/irradiance.py#L2849) to calculate the DNI. Does that solve your problem?

I am not the OP but when I came across this issue - this proposed solution resolved it for me.