oemof/feedinlib #72: Problems while running the feedin function

goncasado6, 2022-05-05:
Hello, after I have done correctly all the previous steps of the weather and setting up the photovoltaic object(which work correctly), when I do the calculate the feedin part it doesnt work, it appears two errors and I dont know why.
Please reply me if you know the error and what I have to do,
Thank you

[image]
[image]
[image]

jesuspolo, 2022-06-08:
I have the same problem, I think that it refers to the King temperature model in pvlib that needs a, b and deltaT parameters, but I'm not able to input these parameters properly to feedinlib

Ludee, 2022-06-16:
Downgrading pvlib to v0.8.1 is an option, but not a good one.

I am currently trying to update an existing pv model to 0.9.1 but I'm having a hard time with different errors.
The documentation and release notes are only helping a bit.
I will post what I found out here.
I also opened a discussion on the [pvlib gGitHub](https://github.com/pvlib/pvlib-python/discussions/1475)

**- temperature model** 
```from pvlib.temperature import TEMPERATURE_MODEL_PARAMETERS
tm_gg = TEMPERATURE_MODEL_PARAMETERS['sapm']['open_rack_glass_glass']
tm_gp = TEMPERATURE_MODEL_PARAMETERS['sapm']['open_rack_glass_polymer']
```

**- PVSystem ** 
```
module_type='glass_glass', # 'glass_polymer'
racking_model='open_rack',
```

Ludee, 2022-06-16:
Yes, I already had a look at the PR.
Not sure if the `temperature_module_parameters` is required or only additional?

GregorBecker, 2022-06-16:
If you set module type and racking model when initializing the PV plant, the temperature models are obtained from the dictionary in pv lib temperature.py. To set these values (a,b, delta T) from the feedinlib is difficult at first sight. 

Without setting the parameters the pv plant cannot be initialized.