oemof/feedinlib #75: Installation complications

FelixMau, 2023-05-18:
I tried to use feedinlib for a Uni project. Unfortunately working versions of `feedinlib` and `windpowerlib` are hard to find. 

My installation from pypi with `pip install feedinlib==0.1.0rc4` failed and using a copy of the `master` branch locally failed due to missing import module `open_FRED`. I cant find it on pypi as well. 
https://github.com/oemof/feedinlib/blob/5e6820a67222b0de93a5ba25514d262f8adbac7a/feedinlib/open_FRED.py#L13

I think the module within feedinlib `open_FRED` should be renamed if it is desired to import another module with the same name within that module.

jonas912, 2023-06-26:
@FelixMau maybe I can help you installing feedinlib. I've also experienced some minor problems getting the package to work which required using some workarounds.  
I'm able to use feedinlib by installing the following packages via pip (I recommend doing it exactly in this order):
`pip install https://github.com/oemof/feedinlib/archive/dev.zip`
`pip install open_FRED-cli`

Alternatively, someone should add `open_FRED-cli` to `install_requires` of setup.py to make the manual installation obsolete.

p-snft, 2023-06-28:
I just merged #76. So, this is solved. Thank you!

closed 2023-06-28 by p-snft