oemof/feedinlib #78: ERA5 data formatting fails with data downloaded from CDS

paulapreuss, 2024-10-09:
`format_windpowerlib` and `format_pvpowerlib` from the `era5.py` module fail with the error `AttributeError: 'DataFrame' object has no attribute 'time'` after trying to format data retrieved from the CDS. The CDS data returned by `get_era5_data_from_datespan_and_position` uses the attribute name `valid_time` instead of `time`. Presumably this was changed with the recent CDS API update.

jenny-nyx, 2025-06-26:
yes, I also believe this was changed with the recent CDS API update. Check out [this pull request](https://github.com/oemof/feedinlib/pull/79) for one potential fix. Alternatively can instead call df["valid_time"] instead of df.time

p-snft, 2025-06-26:
Thanks for referring to the PR. Does it fix the problem in all the cases or is handling of different dataset different? (Please answer in the PR.)